cyassl/cyassl #139: BUILDING_CYASSL is missing in cyassl.vcxproj?

jay, 2015-01-08:
I built the CyaSSL DLL in Visual Studio 2010 using cyassl.vcxproj. To do this I had to add preprocessor define CYASSL_DLL (expected) but also BUILDING_CYASSL. I wonder if that is an oversight or intentional; I'd think you'd want BUILDING_CYASSL regardless.

jay, 2015-01-08:
another thing I had to do was remove `extern` from the `extern __declspec(dllimport/export)` from CYASSL_DLL defines in cyassl/ctaocrypt/visibility.h

ejohnstown, 2015-12-23:
Thank you for the issue report. I'm going to close this since it looks like you provided a pull request fixing this in our current version of this product, [wolfSSL](https://github.com/wolfSSL/wolfssl).

closed 2015-12-23 by ejohnstown