cyassl/cyassl #141: Woes when attempting to support DTLS 1.2 clients with DTLS 1.0 record header version

codeslinger, 2015-02-04:
I have an OpenSSL client using `DTLS_client_method` attempting to connect to a CyaSSL server using `CyaDTLSv1_server_method`. The client sends a `ClientHello` with a DTLS record header version of `0xfeff` (DTLS 1.0) but the `ClientHello.client_version` is `0xfefd` (DTLS 1.2).

The logic in the current `DoClientHello` function does not support the DTLS version numbering scheme. The [minor version comparison therein](https://github.com/cyassl/cyassl/blob/master/src/internal.c#L12799) incorrectly assumes that this is a downgrade attempt and rejects it.

A solution to this might be just to use `CyaDTLSv1_2_server_method`. However, this is prevented by the protocol version check in `GetRecordHeader` [found here](https://github.com/newcolabs/cyassl/blob/biba-master/src/internal.c#L3073). According to my reading of [RFC 5246, Appendix E.1](https://tools.ietf.org/html/rfc5246#appendix-E.1), it would seem that only the `ClientHello.client_version` should matter for version negotiation, not the DTLS record header version. [RFC 6347 section 4.2.1](https://tools.ietf.org/html/rfc6347#section-4.2.1) seems to put an even finer point on it:

> However, in order to avoid the requirement to do version negotiation in the initial handshake, DTLS 1.2 server implementations SHOULD use DTLS version 1.0 regardless of the version of TLS that is expected to be negotiated.  DTLS 1.2 and 1.0 clients MUST use the version solely to indicate packet formatting (which is the same in both DTLS 1.2 and 1.0) and not as part of version negotiation.

CyaSSL servers using `CyaDTLSv1_2_server_method` report DTLS 1.2 in both the DTLS record header and `ServerHello.server_version`. (as of CyaSSL 3.3.2, at least)

How can I support this seemingly RFC-compliant behavior from my clients with a CyaSSL DTLS server? Is there a workaround I can employ?

ejohnstown, 2015-02-05:
Thanks for the notice, I'm looking into this now and will give you an update later. May I ask with what project you are using CyaSSL? (BTW, we are changing the name of the library and have a new repository at https://github.com/wolfssl/wolfssl )

codeslinger, 2015-02-11:
Using it on some internal stuff right now. I just forked CyaSSL; didn't know about the WolfSSL change. I will probably upgrade to that later. I patched CyaSSL to work around this issue internally by only checking the major version for DTLS in `GetRecordHeader`.

ejohnstown, 2015-09-04:
Toby:

I apologize for the delay. This has been fixed in our [wolfSSL repository](https://github.com/wolfSSL/wolfssl) with [commit 58a3656](https://github.com/wolfSSL/wolfssl/commit/58a36566cb06400db69a12855c0c213e7d751969).

--John

closed 2015-09-04 by ejohnstown