cyassl/cyassl #142: unit.test fails

DomT4, 2015-03-05:
Version 3.3.2.

```
make[2]: `testsuite/testsuite.test' is up to date.
make[2]: `tests/unit.test' is up to date.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -j5  check-TESTS
make[2]: warning: -jN forced in submake: disabling jobserver mode.
make[3]: warning: -jN forced in submake: disabling jobserver mode.
PASS: testsuite/testsuite.test
FAIL: tests/unit.test
============================================================================
Testsuite summary for cyassl 3.3.2
============================================================================
# TOTAL: 2
# PASS:  1
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/cyassl/cyassl/issues
============================================================================
make[3]: *** [test-suite.log] Error 1
make[2]: *** [check-TESTS] Error 2
make[1]: *** [check-am] Error 2
make: *** [check] Error 2
```

LOG:

```
starting unit tests...
 Begin API Tests
   CyaSSL_Init(): passed
Client message: hello cyassl!
Server response: I hear you fa shizzle!
Client message: hello cyassl server!
Server response: I hear you fa shizzle!
Client message: hello cyassl server!
Server response: I hear you fa shizzle!
error = -381, Unrecognized host name Error
error = -313, revcd alert fatal error
Client message: hello cyassl server!
Server response: I hear you fa shizzle!
Client message: hello cyassl server!
Server response: I hear you fa shizzle!
   CyaSSL_Cleanup(): passed
 End API Tests
 Begin HASH Tests
   MD5      test passed!
   SHA      test passed!
   SHA-256  test passed!
   SHA-512  test passed!
   SHA-384  test passed!
   RIPEMD   test passed!
   HMAC-MD5 test passed!
   HMAC-SHA test passed!
   HMAC-SHA256 test passed!
   HMAC-SHA384 test passed!
 End HASH Tests
 Begin Cipher Suite Tests
starting default cipher suite tests
notice: using default file tests/test.conf
trying server command line[1]: SuiteTest -v 1 -l ECDHE-ECDSA-CHACHA20-POLY1305 -c ./certs/server-ecc.pem -k ./certs/ecc-key.pem 
trying client command line[1]: SuiteTest -v 1 -l ECDHE-ECDSA-CHACHA20-POLY1305 -A ./certs/server-ecc.pem 
CRL callback url = 
err = -362, CRL missing, not loaded
yassl error: SSL_connect failed
FAIL tests/unit.test (exit status: 1)
```

toddouska, 2015-03-05:
Thanks for the report.  The ecc CRL files in that version expired on Feb 7th.  Please see the new repo: https://github.com/wolfSSL/wolfssl  .  I recommend moving to wolfSSL but you could probably use the certs/ directory from wolfSSL with CyaSSL.

Thanks,
-Todd

DomT4, 2015-03-05:
Ah, thanks! I completely missed the rebrand. Will switch over and see if the issue vanishes. Closing this :).

closed 2015-03-05 by DomT4