cyassl/cyassl #143: Implicit declaration of function 'ForceZero'

davidgfnet, 2015-07-03:
Hello!

I'm having issues compiling cyassl after v3.5.0. I get this message:

    ctaocrypt/src/random.c: In function 'Hash_DRBG_Reseed':
    ctaocrypt/src/random.c:192:5: error: implicit declaration of function 'ForceZero' [-Werror=implicit-function-declaration]
         ForceZero(seed, sizeof(seed));
         ^
    ctaocrypt/src/random.c:192:5: error: nested extern declaration of 'ForceZero' [-Werror=nested-externs]
    ctaocrypt/src/random.c: In function 'RNG_HealthTestLocal':
    ctaocrypt/src/random.c:608:13: error: implicit declaration of function 'ConstantCompare' [-Werror=implicit-function-declaration]
                 if (ConstantCompare(check, outputA, sizeof(check)) != 0)
                 ^
    ctaocrypt/src/random.c:608:13: error: nested extern declaration of 'ConstantCompare' [-Werror=nested-externs]

I'm using gcc 4.9.2, if that matters. With version v3.5.0 it just works OK.

toddouska, 2015-07-08:
Thanks for the report.  The current version of CyaSSL is now wolfSSL : https://github.com/wolfSSL/wolfssl

The commit that is causing problems for you was for our CyaSSL FIPS code that plugs into wolfSSL.  We'll try to resolve that but I do suggest moving to wolfSSL for current features and fixes.

Thanks.

davidgfnet, 2015-07-08:
Hello,
I don't get it, why are both cyassl and wolfssl being mantained then?
I also see common releases... Can you explain it or point me to some docs?
Thanks

toddouska, 2015-07-08:
Short answer, FIPS.  Without CyaSSL FIPS still being used by wolfSSL we wouldn't be changing/maintaining CyaSSL versions and tags (there would just be the final CyaSSL version).  Sorry for the confusion.

davidgfnet, 2015-07-08:
Sorry but I'm not a crypot expert. You mean that cyassl has been authorized
by the US government and wolfssl is still pending?

toddouska, 2015-07-08:
No problem.  Our FIPS submission is based on CyaSSL.  wolfSSL will be submitted this year, so wolfSSL currently uses CyaSSL FIPS.

closed 2015-12-23 by ejohnstown