cyassl/cyassl #149: error build

SergeyDjam, 2018-06-02:
Hello

make check
make -j3  check-am
make[1]: вход в каталог «/home/admin/rpmbuild/BUILD/cyassl-3.12.4»
  CC       ctaocrypt/src/src_libcyassl_la-random.lo
  CC       ctaocrypt/src/src_libcyassl_la-hmac.lo
  CC       ctaocrypt/src/src_libcyassl_la-sha256.lo
ctaocrypt/src/random.c: In function 'Hash_DRBG_Reseed':
ctaocrypt/src/random.c:194:5: warning: implicit declaration of function 'ForceZero' [-Wimplicit-function-declaration]
     ForceZero(seed, sizeof(seed));
     ^
ctaocrypt/src/random.c:194:5: warning: nested extern declaration of 'ForceZero' [-Wnested-externs]
ctaocrypt/src/random.c: In function 'RNG_HealthTestLocal':
ctaocrypt/src/random.c:610:13: warning: implicit declaration of function 'ConstantCompare' [-Wimplicit-function-declaration]
             if (ConstantCompare(check, outputA, sizeof(check)) != 0)
             ^
ctaocrypt/src/random.c:610:13: warning: nested extern declaration of 'ConstantCompare' [-Wnested-externs]
  CC       ctaocrypt/src/src_libcyassl_la-rsa.lo
  CC       ctaocrypt/src/src_libcyassl_la-aes.lo
  CC       ctaocrypt/src/src_libcyassl_la-des3.lo
  CC       ctaocrypt/src/src_libcyassl_la-sha.lo
  CC       ctaocrypt/src/src_libcyassl_la-logging.lo
  CC       ctaocrypt/src/src_libcyassl_la-wc_port.lo
  CC       ctaocrypt/src/src_libcyassl_la-error.lo
  CC       ctaocrypt/src/src_libcyassl_la-memory.lo
  CC       ctaocrypt/src/src_libcyassl_la-asn.lo
  CC       ctaocrypt/src/src_libcyassl_la-coding.lo
  CC       ctaocrypt/src/src_libcyassl_la-arc4.lo
  CC       ctaocrypt/src/src_libcyassl_la-md5.lo
  CC       ctaocrypt/src/src_libcyassl_la-tfm.lo
  CC       src/src_libcyassl_la-internal.lo
  CC       src/src_libcyassl_la-io.lo
  CC       src/src_libcyassl_la-keys.lo
  CC       src/src_libcyassl_la-ssl.lo
  CC       src/src_libcyassl_la-tls.lo
  CC       ctaocrypt/benchmark/benchmark.o
  CC       ctaocrypt/test/test.o
  CC       examples/client/client.o
  CC       examples/server/server.o
  CC       examples/echoclient/echoclient.o
  CC       examples/echoserver/echoserver.o
  CC       ctaocrypt/test/testsuite_testsuite_test-test.o
  CC       examples/client/testsuite_testsuite_test-client.o
  CC       examples/echoclient/testsuite_testsuite_test-echoclient.o
  CC       examples/echoserver/testsuite_testsuite_test-echoserver.o
  CC       examples/server/testsuite_testsuite_test-server.o
  CC       testsuite/testsuite_testsuite_test-testsuite.o
  CC       tests/tests_unit_test-unit.o
  CC       tests/tests_unit_test-api.o
  CC       tests/tests_unit_test-suites.o
  CC       tests/tests_unit_test-hash.o
  CC       examples/client/tests_unit_test-client.o
  CC       examples/server/tests_unit_test-server.o
  CCLD     src/libcyassl.la
  CCLD     ctaocrypt/benchmark/benchmark
  CCLD     ctaocrypt/test/testctaocrypt
  CCLD     examples/client/client
src/.libs/libcyassl.so: undefined reference to `ConstantCompare'
src/.libs/libcyassl.so: undefined reference to `ForceZero'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1584: ctaocrypt/benchmark/benchmark] Ошибка 1
make[1]: *** Ожидание завершения заданий…
src/.libs/libcyassl.so: undefined reference to `ConstantCompare'
src/.libs/libcyassl.so: undefined reference to `ForceZero'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1596: ctaocrypt/test/testctaocrypt] Ошибка 1
src/.libs/libcyassl.so: undefined reference to `ConstantCompare'
src/.libs/libcyassl.so: undefined reference to `ForceZero'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:1608: examples/client/client] Ошибка 1
make[1]: выход из каталога «/home/admin/rpmbuild/BUILD/cyassl-3.12.4»
make: *** [Makefile:2855: check] Ошибка 2

SergeyDjam, 2018-06-03:
I build 3.12.4 version, but

Configuration summary for cyassl version 3.3.2 ?

ejohnstown, 2019-11-11:
Development on CyaSSL has moved to the repository wolfSSL/wolfssl. Please direct all correspondence there. Thank you!

closed 2019-11-11 by ejohnstown